wh201906/Proxmark3GUI #7: unable to connect

ruling, 2020-10-08:
When my PM3 RDV2 uses V0.1.3 firmware and client, the port cannot be connected. When I click to connect, the pop-up dialog box prompts "proxmark3.exe-Cannot find the entry,
Unable to locate the program input point gx_personality_seho and confirm it on the dynamic link library libstdc + +-6.dll". Windows 10 and windows 7 X64 systems all prompt this.
sorry for my poor English！I can only speak Chinese

wh201906, 2020-10-17:
Could you please describe how this issue happens?

wh201906, 2020-11-12:
Seems like the client is not complete.
You can copy the libstdc++-6.dll into the folder where the proxmark3.exe is, then try to use the GUI again.
似乎是你的PM3客户端差一个动态链接库
你可以把libstdc++-6.dll拷贝到proxmark3.exe的所在目录，然后再启动GUI

closed 2020-11-12 by wh201906