wh201906/Proxmark3GUI #9: no member named 'textChanged' in 'QSpinBox'

jsxyhyj, 2020-10-14:
win   qt5.12.8  
Proxmark3GUI-master\ui\mf_trailerdecoderdialog.cpp:18: error: no member named 'textChanged' in 'QSpinBox'

Proxmark3GUI-master\ui\mf_trailerdecoderdialog.cpp:18: error: 'textChanged' is not a member of 'QSpinBox'
     connect(ui->C0Box, &QSpinBox::textChanged, this, &MF_trailerDecoderDialog::on_boxChanged);
                                   ^~~~~~~~~~~

wh201906, 2020-10-17:
Which version are you using? Is it under master branch or dev branch?

wh201906, 2020-11-04:
I know what the problem is. The signal textChanged() is intruduced in Qt 5.14, which will pass the text with prefix and suffix.
I didn't realize the compatibility there. And I will replace it with valueChanged() because I don't need the prefix and suffix.
Thanks for your issue.

commit 6d3dd505 (2020-11-04): Replace textChanged() with valueChanged(). This fixes #9
commit 77d80497 (2020-11-04): Merge pull request #13 from wh201906/master

Replace textChanged() with valueChanged(). This fixes #9
closed 2020-11-04 by wh201906