wh201906/Proxmark3GUI #19: Support v4.13441?

iceman1001, 2021-07-30:
Latest RRG/Iceman release is v4.13441 
https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.13441

wh201906, 2021-08-02:
I‘ll test all features a little bit later.
Kinda busy these days.

wh201906, 2021-09-20:
The V0.2.2 GUI supports RRG v4.13441.
Theoretically, it can support any new clients by changing the config file, which contains the command format.

iceman1001, 2021-09-20:
latest release is  v4.14434  ...

closed 2021-09-20 by iceman1001

wh201906, 2021-09-20:
Yeah, the developing speed of RRG repo is so fast.
I think the V0.2.2 GUI is compatible with v4.14434 client. I haven't test it yet.

wh201906, 2021-09-20:
The command format is loaded from config.json now, so the user can try to change this config file if the command format changes.