wh201906/Proxmark3GUI #22: Raspberry Pi linux 连接不上pm3

lkjopp, 2021-08-12:
连接10次，能连上1次，连接成功显示版本，然后闪掉，log显示

STDIN unexpected end, exit....

wh201906, 2021-08-13:
手上暂时没有PM3硬件，没办法测试
你可以尝试勾选“设置”面板中的“保持所有按钮可点击”（Settings->Keep buttons enabled），然后试一下能否继续使用
或者使用[该版本](https://github.com/RfidResearchGroup/proxmark3/tree/bb0577f97882c6ea7e41d872643626386f173ced)的RRG固件

lkjopp, 2021-08-13:
已测试，在“设置”面板中勾选保持所有按钮可点击，依然无效，点击读卡会跳转原始命令，其他按钮没有反应。

您建议的版本是4.13441，是我第一次使用的版本，这两个版本命令行都可以正常使用。

[image][image]

wh201906, 2021-08-13:
可以把pm3的log文件也发上来吗？

lkjopp, 2021-08-13:
按照你的版本去下载，编译了，刷了pm3，还是不行，GUI连接不上.

lkjopp, 2021-08-13:
pm3的log放在这儿了

https://github.com/lkjopp/log

wh201906, 2021-08-13:
收到，到时候调一下

wh201906, 2021-08-14:
我发现raspbian默认不带qt的开发包，你在编译GUI的时候用的是哪一版Qt？
如果没有用Qt5或者不确定的话可以试下README.md当中的编译命令，我刚才更新过

lkjopp, 2021-08-15:
我安装了qt5，raspberry pi 可以安装的，虽然有些麻烦。

wh201906, 2021-08-15:
颜色应该不影响的，我等手上有PM3硬件之后再试试吧

wh201906, 2021-09-05:
是等待时间和换行符的问题
你可以试一下[这个提交](https://github.com/wh201906/Proxmark3GUI/tree/12a0837c503e3941641640a8ad550811be6f5a82)，我在自己的树莓派上试了下，能用

lkjopp, 2021-09-06:
多谢，晚上回去试完汇报

closed 2021-09-06 by lkjopp

wh201906, 2021-09-06:
后期会跟进的
感谢反馈

wh201906, 2021-09-19:
我忘记了，之前我做过这种情况的处理的
在你之前使用的版本当中勾选“Keep the client active”（在PM3断开后保持客户端运行）就可以了

commit 12a0837c (2021-09-05): Fix a bug

Trying to fix #22
Open client with QProcess::Text for proper newline character(s)
On Raspbian, the isBusy() function will always return false, even the serial port is actually connected.