wh201906/Proxmark3GUI #24: 新iceman客户端不再适用

zzzz0317, 2021-09-17:
新版本iceman客户端似乎改了不少命令

```
[usb|script] pm3 --> hf mf rdbl 0 A FFFFFFFFFFFF
hf mf rdbl: unexpected argument "0"
hf mf rdbl: unexpected argument "A"
hf mf rdbl: unexpected argument "FFFFFFFFFFFF"
[!] Try 'hf mf rdbl --help' for more information.

[usb|script] pm3 --> hf mf rdbl

Read MIFARE Classic block

usage:
    hf mf rdbl [-habv] --blk <dec> [-k <hex>]

options:
    -h, --help                     This help
    --blk <dec>                    block number
    -a                             input key type is key A (def)
    -b                             input key type is key B
    -k, --key <hex>                key, 6 hex bytes
    -v, --verbose                  verbose output

examples/notes:
    hf mf rdbl --blk 0 -k FFFFFFFFFFFF
    hf mf rdbl -b 3 -v             -> get block 3, decode sector trailer
                                  

[usb|script] pm3 --> hf mf rdbl --blk 1 -a -k FFFFFFFFFFFF

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   1 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
```

zzzz0317, 2021-09-17:
#19 重复

closed 2021-09-17 by zzzz0317

wh201906, 2021-09-17:
是的，正在适配中

wh201906, 2021-09-20:
@zzzz0317 已适配