wh201906/Proxmark3GUI #26: 连接不上 pm3

lkjopp, 2021-09-21:
成功make后，报无法找到config.json,发现目录下无此文件，将config_offical.json改为config.json，不再报无法找到config.json，显示已连接一秒，退出，显示未连接。[image]

wh201906, 2021-09-21:
和 #22 的问题一样，你需要在Settings（设置）面板中勾选”Keep the client active even the PM3 hardware is disconnected.(Experimental)“（在PM3断开后保持客户端运行）

wh201906, 2021-09-21:
QSerialPortInfo的isBusy()函数在Raspbian系统中会一直返回false，而我使用这个函数来判断PM3硬件是否断开连接，所以PM3连上之后GUI也会识别为未连接，从而关闭客户端。

lkjopp, 2021-09-21:
试验过了，连接失败，按钮无效，之前#22号中你发的那版，可以正常连接，4.14434也可以连接成功，一些命令可以使用，如果pm3换成老版本可以正常使用。

wh201906, 2021-09-21:
我在自己的Raspbian上面V0.2.2可以配合v4.14434正常使用  
几个要配置的地方：  
1.按照README上的内容编译完成后build文件夹下会有config目录，里面有config_official.json和config_rrgv4.13.json，根据自己使用的客户端类型来选取，并填入Settings->Config file path当中  
2.Settings->Start arguments当中填写"-p <port> -f"（不含引号）
3.Settings面板中勾选”Keep the client active even the PM3 hardware is disconnected.(Experimental)“（在PM3断开后保持客户端运行）
4.最上方Client Path当中填写客户端的可执行文件路径

lkjopp, 2021-09-21:
已成功连接，多谢。

命令json有些需要更新。。

closed 2021-09-21 by lkjopp

wh201906, 2021-09-21:
@lkjopp 有哪些呢？你可以提交一个新issue，或者尝试改好之后提交pull request，我会及时跟进

lkjopp, 2021-09-21:
是我用错了json，很完美，没问题，一切都是那么完美。。。

非常感谢回复