wh201906/Proxmark3GUI #28: MIFARE data display not updated or cleared

Kadalia, 2022-03-12:
Hello,
Nice work ... no ... great work !!
But, I have a minor issues.
The "clear" data button from "File" section is not working. (it works fine for keys). 
The "load" data button doesn't update the Data list display.
v0.2.4 Compiled for ARM
Thank you !

wh201906, 2022-03-12:
> The "clear" data button from "File" section is not working. (it works fine for keys).

Yes I reproduced this bug. I will fix it in the next release. If you need to fix it on your own, open the mainwindow.cpp in src/ui/, find MainWindow::on_MF_File_clearButton_clicked(), then replace the code at line 718

    else if(ui->MF_File_keyButton->isChecked())
with

    else if(ui->MF_File_dataButton->isChecked())

wh201906, 2022-03-12:
> The "load" data button doesn't update the Data list display.

I can't repcoduce this bug. Both the binary data file and the text data file can be loaded on V0.2.4. Did you choose "Data" in the "File" section before loading the data?

Kadalia, 2022-03-12:
Thank you !
Clear Data button is now working.
The Load button also ... hummm ... I was tired :-)
Closing

closed 2022-03-12 by Kadalia

wh201906, 2022-03-12:
@Kadalia Does the "Load" button work after the fix?

Kadalia, 2022-03-12:
> 

Yes :-) I thing it was working before the fix already

wh201906, 2022-03-12:
Ok

Pebrok, 2022-10-16:
Hi, 
I'm still stuck on MAc with the app not connecting to the Proxmark3, even though the terminal app can connect to it. 
I changed the file path as recommanded, but nothing happens. What am I missing? 
[image]
[image]

wh201906, 2022-10-16:
How about checking `Keep the client active even the PM3 hardware is disconnected`

Pebrok, 2022-10-16:
Hi,

thanks for your reply.

I did, but still no connection.

1-Should I change something on the client path, on top? what path should I put? To which file : proxmark3GUI.app or what else ? (I am on Mac)

2-For the config file : should i link to « config_official.json" or to "config_rrgv4.13.json » ?

It does work with terminal, so the connection is working…
It drives me crazy.

Thanks for your help.

[cid:DE5C69D7-9F7B-464A-B663-90966FBE2C95]

***@***.******@***.***>

wh201906, 2022-10-17:
> 1-Should I change something on the client path, on top? what path should I put? To which file : proxmark3GUI.app or what else ? (I am on Mac)

The client path is the same as the command to open proxmark3. If you start proxmark3 in terminal by typing `proxmark3` or `path/to/the/proxmark3`, then you should type it into the `Client Path`

> 2-For the config file : should i link to « config_official.json" or to "config_rrgv4.13.json » ?

The current `config_rrgv4.13` should be fine.

Pebrok, 2022-10-18:
Hi

Still stuck wit no connection of the pm3

I changed what you said.
As I am using terminal, I only put pm3 (the word I used to connect to my Pm3 on terminal)

Any other idea?
Do you see anything missing?
Is there any files I should put in the same folder?

[cid:2FB6BA58-EBD3-49A7-9281-701948403A07]

Pebrok, 2022-10-19:
> Since I don't have a Mac, I can't test what happens. However, I guess this GUI works with v4.14831 on macOS because someone has run it and post the screenshot in README.
> Can you get proxmark3.log in the directory of executable? Does it reports any errors or warnings?

Hi,

and thanks again for your help.

Here is my log from yesterday:
-it doesn't seem to follow the path I'm giving on the setting page (to the config_rrgv4.13.json file).
It is on a path I can't find. (./proxmark3)

[=] Session log /Users/cedricchadapaux/.proxmark3/logs/log_20221018.txt
[=] Output will be flushed after every print.

[+] loaded from JSON file /Users/cedricchadapaux/.proxmark3/preferences.json

[=] Using UART port tty.usbmodemiceman1

[!] ERROR: invalid serial port tty.usbmodemiceman1

[?] Try the shell script `./pm3 --list` to get a list of possible serial ports

[=] Session log /Users/cedricchadapaux/.proxmark3/logs/log_20221018.txt
[+] loaded from JSON file /Users/cedricchadapaux/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1

[=] Session log /Users/cedricchadapaux/.proxmark3/logs/log_20221018.txt
[+] loaded from JSON file /Users/cedricchadapaux/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1

[!!] ERROR: cannot parse command line. We got /dev/tty.usbmodemiceman1 as port and now we got also: /dev/tty.usbmodemiceman1

[=] Session log /Users/cedricchadapaux/.proxmark3/logs/log_20221018.txt
[+] loaded from JSON file /Users/cedricchadapaux/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1

[=] Session log /Users/cedricchadapaux/.proxmark3/logs/log_20221018.txt
[+] loaded from JSON file /Users/cedricchadapaux/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1

wh201906, 2022-10-19:
Could you please clean the log, then run the GUI, then post the new log there?

Pebrok, 2022-10-21:
I am already on V4.14831

Here is the terminal connexion info:

[=] Session log /Users/XXXX/.proxmark3/logs/log_20221021.txt
[+] loaded from JSON file /Users/XXXX/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝
  ╚═╝     ╚═╝     ╚═╝╚════╝     [ Iceman ❄️ ]

Release v4.14831 - Frostbit

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC
I don’t have any proxmark3.log file in my Mac.

wh201906, 2022-10-22:
Can you find the log file at `<somewhere>/.proxmark3/logs/log_<date>.txt`?

Pebrok, 2022-10-22:
Hi,

I finally found the log file (hidden on Mac…)

[+] loaded from JSON file /Users/xxxx/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with............. Clang/LLVM Apple LLVM 13.1.6 (clang-1316.0.21.2)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
       os: RRG/Iceman/master/v4.14831 2022-01-11 19:17:19
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )

wh201906, 2022-10-23:
The log seems fine. Does the GUI shows `Connected` at first, then changes to `Disconnected`?

Pebrok, 2022-10-23:
Not, it has never shown "connected".

Do you think I should change the path to the driver , same as the one the proxmark3 uses, and to the Json file?
Or else?

xxx

wh201906, 2022-10-24:
I think the log shows the client works properly. I don't know what else can be wrong.