wh201906/Proxmark3GUI #30: "Failed to Load config" on linux with the head of iceman proxmark3

gbin, 2022-05-13:
First I had to work around the incorrect CLI syntax for the client:
-p /dev/<port> -f

It looks like it starts but then I don't see the reason for the failure in the captured log:
```

[=] Session log /home/gbin/.proxmark3/logs/log_20220513.txt
[=] Output will be flushed after every print.

[+] loaded from JSON file /home/gbin/.proxmark3/preferences.json

[=] Using UART port /dev/ttyACM0

[=] Communicating with PM3 over USB-CDC

[usb|script] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]

  Iceman/master/v4.14831-597-gdb2157661 2022-05-13 08:27:57 fcb68954e

  compiled with............. GCC 12.1.0

  platform.................. Linux / x86_64

  Readline support.......... present

  QT GUI support............ present

  native BT support......... present

  Python script support..... present

  Lua SWIG support.......... present

  Python SWIG support....... present

 [ PROXMARK3 ]

  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14831-597-gdb2157661 2022-05-13 08:27:55 fcb68954e
       os: Iceman/master/v4.14831-597-gdb2157661 2022-05-13 08:28:00 fcb68954e
  compiled with GCC 12.1.0

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 58% used )
```

wh201906, 2022-05-14:
Hi. You need to follow these steps
1. Go to the Settings panel, set the start argument to `-p /dev/<port> -f`. The client will remember these settings.
1. Set the config file path, like `../config/config_rrgv4.13.json`, then connect
2. If the GUI still can't connected to the hardware, check the `Keep the client active even the PM3 hardware is disconnected`

gbin, 2022-05-14:
Thanks it works! maybe having some kind of sane defaults might help the usability :)

wh201906, 2022-05-14:
OK
Thanks for your advice.

closed 2022-05-15 by wh201906