wh201906/Proxmark3GUI #36: 为什么刷了冰人固件用这个gui读不出卡信息了？

Mabisi, 2022-10-11:
ic卡读取 什么也没有返回，id卡正常，请问这个是什么原因呢？

Mabisi, 2022-10-12:
V0.2.5-win64-rrg_other-v4.14831

匹配，用的 rrg 14831这个版本

wh201906, 2022-10-13:
要不先试一下v4.14434？目前我手上没有设备，等过几周我会去试一下v4.14831有什么问题

Mabisi, 2022-10-14:
三个版本都试了，全都无法读取高频卡。。然后我又刷回以前买的软件的固件发现也不能读取了。。。啥情况大佬

wh201906, 2022-10-14:
会不会是硬件有问题？在命令行下面跑hw tune看一下两个天线能正常工作吗

Mabisi, 2022-10-14:
测试了是正常的😢

`[usb|script] pm3 --> hw status

[#] Memory

[#]   BigBuf_size............. 44520

[#]   Available memory........ 44520
[#] Tracing

[#]   tracing ................ 1

[#]   traceLen ............... 0

[#] Current FPGA image

[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] LF Sampling config

[#]   [q] divisor............. 95 ( 125.00 kHz )

[#]   [b] bits per sample..... 8

[#]   [d] decimation.......... 1

[#]   [a] averaging........... yes

[#]   [t] trigger threshold... 0

[#]   [s] samples to skip..... 0 

[#] 

[#] LF T55XX config

[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]

[#]            mode            |start|write|write|write| read|write|write

[#]                            | gap | gap |  0  |  1  | gap |  2  |  3

[#] ---------------------------+-----+-----+-----+-----+-----+-----+------

[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 

[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 

[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 

[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 

[#] HF 14a config

[#]   [a] Anticol override.... std    ( follow standard )

[#]   [b] BCC override........ std    ( follow standard )

[#]   [2] CL2 override........ std    ( follow standard )

[#]   [3] CL3 override........ std    ( follow standard )

[#]   [r] RATS override....... std    ( follow standard )

[#] Transfer Speed

[#]   Sending packets to client...

[#]   Time elapsed................... 500ms

[#]   Bytes transferred.............. 287232

[#]   Transfer Speed PM3 -> Client... 574464 bytes/s

[#] Various

[#]   Max stack usage......... 4120 / 8480 bytes

[#]   Debug log level......... 1 ( error )

[#]   ToSendMax............... -1

[#]   ToSend BUFFERSIZE....... 2308

[#]   Slow clock.............. 31461 Hz

[#] Installed StandAlone Mode

[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)

[#] 

[usb|script] pm3 --> hw tune

[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
[\] 20
[|] 19
[/] 18
[-] 17
[\] 16
[|] 15
[/] 14
[-] 13
[\] 12
[|] 11
[/] 10
[-]  9

[=] ---------- LF Antenna ----------
[+] LF antenna: 26.30 V - 125.00 kHz
[+] LF antenna: 22.32 V - 134.83 kHz
[+] LF optimal: 27.71 V - 127.66 kHz
[+] Approx. Q factor (*): 7.4 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 14.84 V - 13.56 MHz
[+] Approx. Q factor (*): 4.3 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.`

wh201906, 2022-10-14:
那么命令行读卡正常吗？

Mabisi, 2022-10-14:
[usb|script] pm3 --> hf 14a info

[usb|script] pm3 --> hf 14a info -v

[!] iso14443a card select failed

依旧无法读取.

wh201906, 2022-10-14:
你把卡片放到高频天线上，然后跑hw tune，看一下有没有压降？

Mabisi, 2022-10-14:
有一点。。

[usb|script] pm3 --> hw tune

[=] ---------- LF Antenna ----------
[+] LF antenna: 26.30 V - 125.00 kHz
[+] LF antenna: 22.27 V - 134.83 kHz
[+] LF optimal: 27.67 V - 127.66 kHz
[+] Approx. Q factor (*): 7.2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 13.64 V - 13.56 MHz
[+] Approx. Q factor (*): 4.0 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

[usb|script] pm3 --> hw tune

[=] ---------- LF Antenna ----------
[+] LF antenna: 26.27 V - 125.00 kHz
[+] LF antenna: 22.27 V - 134.83 kHz
[+] LF optimal: 27.66 V - 127.66 kHz
[+] Approx. Q factor (*): 7.2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 14.80 V - 13.56 MHz
[+] Approx. Q factor (*): 4.3 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

wh201906, 2022-10-14:
基本没有压降
应该是硬件问题，要么卡片有问题要么PM3有问题

Mabisi, 2022-10-14:
感谢大佬，那我在研究研究，固件刚入手的新的刚用了几次，换冰人固件之前都是正常的。。。

wh201906, 2022-10-25:
我目前手上的硬件跑hw tune空载是28.3V，有卡的时候是26.5V。感觉还是你的硬件有问题，电压不应该只有10多伏的
这个issue我先关闭了，如果有进一步的发现你可以再打开

closed 2022-10-25 by wh201906