wh201906/Proxmark3GUI #38: 图形化没有支持一些新命令

xbebhxx3, 2022-11-27:
但是图形化没有支持一些新命令，比如auto、hf mf autopwn
命令列表在这里[https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/commands.md](url)

wh201906, 2022-11-27:
因为我刚开始设计GUI的时候是针对官方客户端设计的，之后才适配了冰人客户端，所以只做了官方客户端和冰人客户端都支持的命令。考虑到`hf mf autopwn`的适用场景比较广，我后期应该会添加对这个命令的支持

schmiernippel, 2023-01-17:
Please add a button for hf mf autopwn...this would make things really easy :-)

wh201906, 2023-01-17:
> Please add a button for hf mf autopwn...this would make things really easy :-)

I will add it in the next release, but I'm kinda busy these days.

schmiernippel, 2023-01-17:
Great....thanks!!!

onepercentLI, 2023-01-25:
大佬考虑下加入字典爆破功能，\client\dictionaries目录下有不少字典文件，通过命令
`pm3 --> hf mf chk --1k -f mfc_default_keys`可以一个文件一个文件爆破，但是这样有点麻烦。

wh201906, 2023-01-25:
> 大佬考虑下加入字典爆破功能，\client\dictionaries目录下有不少字典文件，通过命令 `pm3 --> hf mf chk --1k -f mfc_default_keys`可以一个文件一个文件爆破，但是这样有点麻烦。

估计过几个版本会把这个功能加进去

schmiernippel, 2023-02-16:
> > Please add a button for hf mf autopwn...this would make things really easy :-)
> 
> I will add it in the next release, but I'm kinda busy these days.

Any news when the next release comes? :-)

wh201906, 2023-02-17:
> Any news when the next release comes? :-)

I’m not sure...

schmiernippel, 2023-02-17:
> > Any news when the next release comes? :-)
> 
> I’m not sure...

Ok, thanks anyway for your great work!!!

AlexMicrotech, 2023-03-03:
WIN32 версия ошибается на Windows 8 32 bit, "Failed to start the client"

wh201906, 2023-03-03:
@AlexMicrotech 
Could you please create a new issue for this?