wh201906/Proxmark3GUI #41: Client connects to PM3 for a second then disconnects

danielbercu, 2022-12-17:
Hi, I managed to correctly build the client and link the port and the actual CLI client that I use to connect to my Proxmark 3 X (https://shop.mtoolstec.com/product/proxmark3-x), but after I hit "Connect" with the correct port selected it initializes the connection but then instantly disconnects.
Here is what happens on the terminal:
`$ ./Proxmark3GUI 
QMetaObject::connectSlotsByName: No matching signal for on_Raw_keyPressed(QObject*,QEvent&)
QMetaObject::connectSlotsByName: No matching signal for on_MF_keyWidget_resized(QObject*,QEvent&)
QMetaObject::connectSlotsByName: No matching signal for on_portSearchTimer_timeout()
QMetaObject::connectSlotsByName: No matching signal for on_GroupBox_clicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_stopButton_clicked()
dock count 5
dock name "Mifare"
dock name "LF"
dock name "T55xx"
dock name "RawCommand"
dock name "Settings"
Main: QThread(0x55aa5db463c0)
"/home/daniel/Proxmark3GUI/build"
"/home/daniel/Proxmark3GUI/build"
"/home/daniel/Proxmark3GUI/build"
config file: ":/config/config_official.json"
QThread(0x55aa5df0d530)
PM3 Error: QProcess::Crashed "Process crashed"
`

wh201906, 2022-12-17:
You might need to check "Keep the client active even the PM3 hardware is disconnected." in the "Settings" panel
[image]

danielbercu, 2022-12-17:
That solved it, thanks!

closed 2022-12-17 by danielbercu

vincent-stadler, 2023-01-25:
Solved the issue for me as well. Thanks!