wh201906/Proxmark3GUI #42: can not connect to pm3 | 应用程序无法正常启动

PikuZheng, 2022-12-21:
[image]

感觉是缺少依赖库，不过缺少的是什么呢

wh201906, 2022-12-21:
你想要打开GUI还是命令行的客户端？
打开GUI需要点击压缩包内GUI文件夹下的Proxmark3GUI.exe
打开命令行需要点开压缩包内的Go.bat

另外，官方版（V0.2.6-win64-official-v3.1.0）已经很久没有更新了，如果会刷固件建议使用最新的冰人版（V0.2.6-win64-rrg_other-v4.15864）

PikuZheng, 2022-12-21:
感谢回复，都试了，拿process monitor跟踪了一下 仿佛是设置-预加载脚本路径 那个bat没有起作用，没有从lib文件夹找dll文件。将lib文件夹所有文件复制到pm3程序文件夹后可以正确运行

wh201906, 2022-12-21:
似乎确实有问题

wh201906, 2022-12-21:
所有测试均基于`V0.2.6-win64-official-v3.1.0.7z`
> 仿佛是设置-预加载脚本路径 那个bat没有起作用，没有从lib文件夹找dll文件。

应该不是这个问题。我直接在压缩包内打开Proxmark3GUI.exe，然后用火绒剑分析proxmark3.exe，发现所需的dll文件是从lib文件夹加载的
[image]

会不会是因为你的固件不是官方版？如果官方版的客户端连不上设备，那么GUI的原始命令当中就没有输出
我把手上的设备烧好官方固件之后测下来可以正常使用

closed 2022-12-21 by wh201906

wh201906, 2022-12-21:
> 将lib文件夹所有文件复制到pm3程序文件夹后可以正确运行

如果预加载脚本确实有问题，这也是个解决办法。