wh201906/Proxmark3GUI #48: Help

sparespice, 2023-07-12:
This is the message I get after copy and pasting the build code. Is there something I need downloaded before doing this? I already set up my Prox and its working fine with the Iceman build. I just wanted to try this out. 

Message:
Package qt5-default is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'qt5-default' has no installation candidate
Cloning into 'Proxmark3GUI'...
remote: Enumerating objects: 517, done.
remote: Counting objects: 100% (517/517), done.
remote: Compressing objects: 100% (498/498), done.
remote: Total 517 (delta 22), reused 468 (delta 13), pack-reused 0
Receiving objects: 100% (517/517), 3.59 MiB | 1.45 MiB/s, done.
Resolving deltas: 100% (22/22), done.
Info: creating stash file /home/kali/Proxmark3GUI/build/.qmake.stash
Project ERROR: Unknown module(s) in QT: serialport
make: *** No targets specified and no makefile found.  Stop.
zsh: no such file or directory: ./Proxmark3GUI

wh201906, 2023-07-13:
What OS are you using?

wh201906, 2023-07-13:
These links might be helpful for you.

https://www.geeksforgeeks.org/how-to-enable-universe-repository-in-ubuntu/
https://stackoverflow.com/questions/74110674/problem-installing-qt5-default-on-ubuntu-22-04

sparespice, 2023-07-13:
I use Linux..hopefully these work, thank you!

sparespice, 2023-07-13:
Yeah idk man its still not working for me. This the commands the websites gave me and i'm still getting errors.
 ─$ sudo apt install qtbase5-dev qt5-qmake
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
qtbase5-dev is already the newest version (5.15.8+dfsg-12).
qt5-qmake is already the newest version (5.15.8+dfsg-12).
qt5-qmake set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 517 not upgraded.
                                                                                                            
└─$ sudo apt-get install qt5-default  
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package qt5-default is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'qt5-default' has no installation candidate

wh201906, 2023-07-13:
I guess you don't need to install qt5-default then.

wh201906, 2023-07-13:
Please try these
```
sudo apt-get install libqt5serialport5-dev
git clone https://github.com/wh201906/Proxmark3GUI.git --depth=1
cd Proxmark3GUI
mkdir build && cd build
qmake ../src
make -j4 && make clean
./Proxmark3GUI

```

sparespice, 2023-07-15:
Ok now that I did that I was able to get into the ProxmarkGUI but now my Proxmark3 easy wont connect. I follow the instructions from the Youtube video and could not connect. This is the message it gives me.

 $ ./Proxmark3GUI
QMetaObject::connectSlotsByName: No matching signal for on_Raw_keyPressed(QObject*,QEvent&)
QMetaObject::connectSlotsByName: No matching signal for on_MF_keyWidget_resized(QObject*,QEvent&)
QMetaObject::connectSlotsByName: No matching signal for on_portSearchTimer_timeout()
QMetaObject::connectSlotsByName: No matching signal for on_GroupBox_clicked(bool)
QMetaObject::connectSlotsByName: No matching signal for on_stopButton_clicked()
dock count 5
dock name "Mifare"
dock name "LF"
dock name "T55xx"
dock name "RawCommand"
dock name "Settings"
Main: QThread(0x5590d31800)
port: "ttyACM0"
"/home/kali/Proxmark3GUI/build"
"/home/kali/Proxmark3GUI/build"
"/home/kali/Proxmark3GUI/data"
config file: ":/config/config_rrgv4.15864.json"
unexpected output: "[!!] ERROR: invalid parameter: -p/dev/ttyACM0\n\n"
PM3 Error: QProcess::Crashed "Process crashed"

wh201906, 2023-07-15:
The arguments in the Settings panel should be `-p /dev/<port> -f` or `-p <port> -f`
By the way, what is the youtube video you mentioned above?

sparespice, 2023-07-31:
https://youtu.be/iwqyE5dboeo this the video

wh201906, 2023-07-31:
> The arguments in the Settings panel should be `-p /dev/<port> -f` or `-p <port> -f`

Does these arguments work?

sparespice, 2023-07-31:
No I think it’s just my proxmark bc it’s errors in the hw status when I’m in pm3. I just hoped back on this project today.