wh201906/Proxmark3GUI #51: 冰人固件+HC-05蓝牙无法连接

Otakukisaragishintaro, 2023-08-27:
是冰人固件本来就连接不了蓝牙，还是没完善好？

wh201906, 2023-08-27:
冰人固件可以连蓝牙的
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md
你用的是RDV4吗？

Otakukisaragishintaro, 2023-08-30:
是 512的easy改的

wh201906, 2023-08-30:
你是怎么改的呢？把HC-05飞线到MCU的USART1对应的引脚上吗？

Otakukisaragishintaro, 2023-09-01:
把tx和rx飞到对应的pa21和pa22 （11脚和14脚）也有在用蓝牙的上板

wh201906, 2023-09-01:
你编译固件的时候加上PLATFORM_EXTRAS=BTADDON了吗？

Otakukisaragishintaro, 2023-09-01:
我直接刷的 没有编译

wh201906, 2023-09-01:
直接刷的Easy固件不带蓝牙功能，你需要改参数之后自己编译一下
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md
如果电脑上没有对应的开发环境可以试试这个仓库，用GitHub Action来按照需求编译冰人(RRG)固件
https://github.com/wh201906/PM3-firmware-builder

Otakukisaragishintaro, 2023-09-02:
我是用你的那个gui直接刷的冰人固件，也就是说我得重新编译？但是我不会啊……

 

 https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md
 https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md
 https://github.com/wh201906/PM3-firmware-builder

wh201906, 2023-09-02:
我这边倒是可以帮你编译一个
你做了哪些魔改？只有HC-05吗？还是说有外挂Flash之类其它的

Otakukisaragishintaro, 2023-09-02:
只有hc05和电池

wh201906, 2023-09-02:
你要哪个版本的固件？v4.16717吗？

Otakukisaragishintaro, 2023-09-02:
就现在最新的就行，谢谢大佬

wh201906, 2023-09-02:
https://github.com/wh201906/PM3-firmware-builder/actions/runs/6056475519
下面那个Artifacts就是编译好的固件，v4.16717是最新的发行版，master是最新的开发版，你可以都试一下

Otakukisaragishintaro, 2023-09-02:
谢谢大佬 我试试

 
https://github.com/wh201906/PM3-firmware-builder/actions/runs/6056475519

Otakukisaragishintaro, 2023-09-02:
提示ARM firmware does not match the source at the time the client was compiled
 

865187685
***@***.***

 
https://github.com/wh201906/PM3-firmware-builder/actions/runs/6056475519

Otakukisaragishintaro, 2023-09-02:
[=] Session log D:\V0.2.8-win64-rrg_other-v4.16717\client\/.proxmark3/logs/log_20230902.txt
[+] loaded from JSON file D:\V0.2.8-win64-rrg_other-v4.16717\client\/.proxmark3/preferences.json
[+] About to use the following file:
[+]    bootrom.elf
[+] Loading ELF file bootrom.elf
[+] ELF file version Iceman/master/32f892e-suspect 2023-09-02 05:05:21 3c525426b
[!] ARM firmware does not match the source at the time the client was compiled
[=] Make sure to flash a correct and up-to-date version
[=] You can force flashing this firmware by using the option '--force'
[!!] Aborted on error

[=] Have a nice day!

请按任意键继续. . .

865187685
***@***.***

 

865187685

 

 
https://github.com/wh201906/PM3-firmware-builder/actions/runs/6056475519

wh201906, 2023-09-02:
编辑bat文件，加上--force选项就可以了
固件版本而和客户端版本不一定要精确匹配，差距不大就行

Otakukisaragishintaro, 2023-09-02:
试了一下，还是不行，蓝牙模块波特率我也改了115200，不知道问题出在哪里

865187685
***@***.***

Otakukisaragishintaro, 2023-09-02:
[+] loaded from JSON file D:\V0.2.8-win64-rrg_other-v4.16717\client\/.proxmark3/preferences.json
 
[=] Using UART port COM268
 
[=] Communicating with PM3 over FPC UART
 
[=] PM3 UART serial baudrate: 115200
 

 
[=] No previous history could be loaded
 
[fpc|script] pm3 --&gt; hw version
 

 
 [ Proxmark3 RFID instrument ]
 

 
 [ CLIENT ]
 
  Iceman/master/v4.16717-9-gadfebd651 2023-06-26 21:52:22 c446c2de4
 
  compiled with............. MinGW-w64 10.3.0
 
  platform.................. Windows (64b) / x86_64
 
  Readline support.......... present
 
  QT GUI support............ present
 
  native BT support......... absent
 
  Python script support..... absent
 
  Lua SWIG support.......... present
 
  Python SWIG support....... absent
 

 
 [ PROXMARK3 ]
 
[!] Received packet frame with variable part too short? 313/455
 
[!] Received packet OLD frame with payload too short? 134/534

 

865187685
***@***.***

wh201906, 2023-09-02:
首先是烧固件的时候需要改烧固件的bat脚本，在`bash`开头的那一行尾部加上--force选项，这样才能把版本不匹配的固件刷进去
其次，你的蓝牙模块也改了115200的波特率吗？PM3的默认波特率是115200，但是HC-05的数据默认波特率好像是9600
另外你的Tx和Rx连线的时候有交叉线序吗？

Otakukisaragishintaro, 2023-09-02:
你说的这些，我都做了，不加我后面也烧录不进去，蓝牙模块肯定也改了，然后TX和RX也是交叉的 

865187685
***@***.***

wh201906, 2023-09-02:
那我也不知道原因了。或许你可以去pm3的官方Discord服务器里面问问