wh201906/Proxmark3GUI #52: 关于hardnested算nonces.bin

jiajixin521, 2023-09-14:
计算nonces.bin在有些电脑上会显示
`Couldn't read benchmark data. Assuming brute force rate of 120000000 states per second`
请问这种情况一般是哪方面的问题。
12700h和7700h，12900h都试过，系统都是win11 22h2

wh201906, 2023-09-14:
你的客户端下的`resources/`里面有`hardnested_bf_bench_data.bin`这个文件吗？从代码上来看测试跑分需要用到这个文件
https://github.com/RfidResearchGroup/proxmark3/blob/25ab50713751d03424a8d4c6c76a5bb34d90769e/client/deps/hardnested/hardnested_bruteforce.c#L494
https://github.com/RfidResearchGroup/proxmark3/blob/25ab50713751d03424a8d4c6c76a5bb34d90769e/client/deps/hardnested/hardnested_bruteforce.c#L406

jiajixin521, 2023-09-14:
我这边下载的是0.28win64official版本，这个文件是有的但目录在`V0.2.8-win64-official-v3.1.0\client\hardnested`下面

jiajixin521, 2023-09-14:
抱歉。我犯了一个极其愚蠢的错误，我把他放在桌面执行了，应该是权限不够导致的找不到benchmark data，实在抱歉打扰了。

jiajixin521, 2023-09-14:
另外，感谢指导。

wh201906, 2023-09-14:
不客气，那我就先把issue关了

closed 2023-09-14 by wh201906