wh201906/Proxmark3GUI #53: Connection on ubuntu 22.04 to device not working

setime, 2023-10-28:
Issue: 
Trying to connect to a device on Ubuntu 22.04 failes with the error message: 
```
unexpected output: "Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.\n"
```

Steps to reproduce: 
* Execute the GUI
* Push the `Connect` button 

Preliminary analysis of the issue: 
Using wayland promts in an additional message from the client. This is not expected from the GUI and breaks the connection. The issue is in the function `PM3Process::connectPM3`. Setting a breakpoint in the function is a work around, because then reading the output from the command line is delayed and the string `os: ` is available in the result variable. 

Proposed fix:  
If the error message is in the `result` variable wait long until the expected string is available in the result variable.

setime, 2023-10-28:
Ugly workaround: 

```
From ae9775c2ae929f5d7546d2dcbe24dd8cae878080 Mon Sep 17 00:00:00 2001
From: timeframe <timeframe1@gmx.de>
Date: Sat, 28 Oct 2023 10:15:06 +0200
Subject: [PATCH] Quick workaround to get the GUI working on Ubuntu 22.04

---
 src/common/pm3process.cpp | 1 +
 1 file changed, 1 insertion(+)

diff --git a/src/common/pm3process.cpp b/src/common/pm3process.cpp
index 76c13b8..9443421 100644
--- a/src/common/pm3process.cpp
+++ b/src/common/pm3process.cpp
@@ -32,6 +32,7 @@ void PM3Process::connectPM3(const QString& path, const QStringList args)
     start(path, args, QProcess::Unbuffered | QProcess::ReadWrite);
     if(waitForStarted(10000))
     {
+        QThread::msleep(5000);
         waitForReadyRead(10000);
         setRequiringOutput(false);
         result = *requiredOutput;
-- 
2.34.1

```

wh201906, 2023-10-28:
> Preliminary analysis of the issue:
Using wayland promts in an additional message from the client. This is not expected from the GUI and breaks the connection. The issue is in the function PM3Process::connectPM3. Setting a breakpoint in the function is a work around, because then reading the output from the command line is delayed and the string os: is available in the result variable.

> Proposed fix:
If the error message is in the result variable wait long until the expected string is available in the result variable.

Thank you for providing me these!

wh201906, 2023-10-28:
Would you like to make a PR for your workaround? I might implement the full fix you mentioned later.

setime, 2023-10-28:
Sure I can make a PR for this workaround. It delays the connection time, but it works.

wh201906, 2023-10-28:
That helps. Thanks!

setime, 2023-10-28:
PR is created to the `dev` branch. Just tell me if you want a PR to the master.

wh201906, 2023-10-28:
I've merged that. Thanks!

closed 2023-10-28 by setime

setime, 2023-10-29:
FYI: For anyone who wonders, the issue arises with the `https://github.com/Proxmark/proxmark3` repo.