wh201906/Proxmark3GUI #55: Ubuntu 22.04 and Iceman Fork does not work

setime, 2023-10-29:
Issue: 

Using the iceman proxmark3 fork (https://github.com/RfidResearchGroup/proxmark3) results in the following error. 

```
[=] Session log /home/setime/.proxmark3/logs/log_20231029063100.txt
[+] loaded from JSON file `/home/setime/.proxmark3/preferences.json`
[=] Using UART port /dev/ttyACM0
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
```

Logfile: 
```
[+] loaded from JSON file `/home/setime/.proxmark3/preferences.json`
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
[!!] STDIN unexpected end, exit...
```

Working the Iceman installation via the terminal works without any issues. Termial logs:

```
> /home/setime/repos/proxmark3_iceman/pm3 -p /dev/ttyACM0
[=] Session log /home/setime/.proxmark3/logs/log_20231029061950.txt
[+] loaded from JSON file `/home/setime/.proxmark3/preferences.json`
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 61% used )

    Client.... Iceman/master/v4.17140-208-g9fd539723 2023-10-28 21:24:05
    Bootrom... Iceman/master/v4.17140-208-g9fd539723-suspect 2023-10-28 21:23:45 
    OS........ Iceman/master/v4.17140-208-g9fd539723-suspect 2023-10-28 21:24:01 
    Target.... PM3 GENERIC

[usb] pm3 -> 
```

The output from the GUI does seem to stop after the wayland error. 

Analysis:
I was unable to find the issue. It seems that the QProcess input somhow screws things up.

wh201906, 2023-10-29:
So it only works with the official client, right?

setime, 2023-10-29:
Yes, I had it working with the official client, after my fix. Then I reflashed my device and tried switching to the iceman client and this didnt work, I tried the GUI before my fix and got the same result.

wh201906, 2023-11-03:
I've tested GUI commit 38c7916 and e2fb189 with the same client commit as yours ([9fd539723](https://github.com/RfidResearchGroup/proxmark3/tree/9fd539723ccd574e024fe2286ec289d9bad24055)), but I can't reproduce your result.

38c7916
[image]
e2fb189
[image]