wh201906/Proxmark3GUI #59: win10下找不到dll文件

wocaishiniliu, 2024-04-26:
在图形化界面打开报错如下：
[image]

打开client下的procmark3.exe报错如下：一共三个
[image]
[image]
[image]

wh201906, 2024-04-26:
你看看你的系统盘下Windows/system32/目录里面有没有个zlib开头的dll文件

wocaishiniliu, 2024-04-26:
> 你看看你的系统盘下Windows/system32/目录里面有没有个zlib开头的dll文件

您说的是这个么：
[image]

wh201906, 2024-04-26:
> 您说的是这个么： ![1714121167448](https://private-user-images.githubusercontent.com/57513031/325907838-3ccda1ae-d7b2-43e5-97cb-0cfdc0d1d462.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxMjMxNTgsIm5iZiI6MTcxNDEyMjg1OCwicGF0aCI6Ii81NzUxMzAzMS8zMjU5MDc4MzgtM2NjZGExYWUtZDdiMi00M2U1LTk3Y2ItMGNmZGMwZDFkNDYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI2VDA5MTQxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1NjM2YWE4ZGJmMzJjYzYyNTRmMDk4MGI1NGQ4YjVmOGZhZTEzZjJkNjE3OWM3YTUyNTMzYjhhY2U2ZWQ1Y2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G9oHgU7wt1E6y4Fdl0c2CBVzhHkO29NdR_9vjugOPeM)

是这个。这个问题我没记错的话是由于intel的wifi驱动导致的。我过几天会给你发一个zlib1.dll，把它放到GUI所在目录即可

wh201906, 2024-04-26:
如果方便的话你可以把你的system32文件夹下的这个zlib1.dll发过来，我方便复现

wocaishiniliu, 2024-04-26:
好的，非常感谢！！！
我的dll是这个
[zlib1.zip](https://github.com/wh201906/Proxmark3GUI/files/15128217/zlib1.zip)

wh201906, 2024-04-28:
你试试看把`client\libs\`目录下的zlib1.dll复制到`client\`目录下，然后再从GUI里面启动试试

wocaishiniliu, 2024-04-28:
GUI不报图片上的错了，但是依旧无法连接。显示如下：
[image]

我试了重新刷了固件，无法解决。固件是刷成功了。另一个GUI软件可以打开，显示内容如下：
[image]

尝试从client里面的exe启动，仍然报上面三个图片的错。
多谢帮忙调试！！！

wh201906, 2024-04-28:
你刷的是什么固件？从你发的GUI图上来看你正在使用的固件版本太老了

wocaishiniliu, 2024-04-29:
我是刷入的这个项目最新的release下载的压缩包中的固件。见下面截图：
[image]
您能提供较新的固件的链接么？我确实不太清楚该选用哪个固件。多谢！！

wh201906, 2024-04-29:
有两个问题：
1. 固件分为两部分，一部分是bootrom，一部分是fullimage，从上一张截图上来看bootrom是刷进去了，但是fullimage没刷
2. 从上一张截图上来看你的PM3是256kB的版本而不是512kB的版本，装不下最新的冰人固件，你只能试下比较老的官方固件（2018年10月）或者是关闭一部分冰人固件的功能以后重新编译源码以缩小尺寸

wocaishiniliu, 2024-04-30:
太强了！！
您提的两个问题，归结到PM3的存储太小。我想到两个解决方案。1. 重新编译冰人固件，我试试，还没玩过。2. 给PM3换一个更大的存储，但是限于知识边界，希望您可以评估可行性。如下是我PM3的拍照。
[image]

closed 2024-04-30 by wocaishiniliu

wh201906, 2024-04-30:
> 1. 重新编译冰人固件

你可以参考一下这个[文档](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#256KB-versions)，里面有写怎么关闭不需要的功能。

> 2. 给PM3换一个更大的存储

理论上是可行的，但是需要一定的焊接功底。图上写着AT91SAM7S256的芯片就是PM3用的MCU，自带256kB存储。你需要把它取下来，换成AT91SAM7S512，然后从旁边的JTAG口刷一次固件进去，这样就换好了