wh201906/Proxmark3GUI #60: Could not recognize the device (请求 USB 设备描述符失败)

wocaishiniliu, 2024-06-09:
根据之前请教的结果，更换了更大的存储，从256KB换到了512KB。
目前win10计算机无法识别设备，如下图。
[image]
PM3设备上所有灯常亮，Button没有反应。
尝试安装驱动，失败，截图如下。
[image]
现在没有太好的思路调试。一个想看看电路图，如果换了内存芯片，电阻有改变，需不需要更换某个电阻。另一个怀疑还是驱动的问题。请求指导和建议

wocaishiniliu, 2024-06-09:
之前的issue在   
https://github.com/wh201906/Proxmark3GUI/issues/59

wh201906, 2024-06-09:
你通过JTAG脚烧录过固件吗？
你换掉的不是内存芯片，而是一整个MCU。因为新MCU里面的固件不是Proxmark3的固件，所以甚至可能没有配置USB外设，因此电脑读不到，也无法通过USB烧录固件进去，只能通过调试接口（JTAG）先烧一次固件才能继续使用。

wocaishiniliu, 2024-06-13:
在尝试通过JTAG脚烧录过固件的时候遇到了困难，网上说的解决方案可以通过ISP使用串口线进行烧录，也可以使用Jlink v8进行烧录。整个流程一点不懂，能否提供一个 step by step的视频或者博客，多谢！！！

wh201906, 2024-06-13:
我没有尝试过用ISP方式烧写固件，但似乎是可行的。你可以参考[这个网站](https://startingelectronics.org/tutorials/AT91SAM7S-microcontrollers/programming-AT9SAM7S-EK-using-SAM-BA/)上的步骤试试看
至于用jtag烧写，任何支持jtag的调试器都可以烧进去，包括但不限于jlink。你可以参考下[Proxmark3仓库里的教程](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md)。大体思路是将调试器和板子的jtag口连接起来，然后用OpenOCD连上调试器之后检查能否扫描到MCU，如果能扫描到的话将client目录下的recovery/proxmark3_recovery.bin烧入即可。