wh201906/Proxmark3GUI #61: MACOS连接问题客户端路径无效

584146519, 2024-07-26:
[image]
[image]
试了很多路径，包括教程的都不行，不知道路径该是什么样子的

dimovic69, 2024-09-19:
Same problem here...

Y1ng0, 2024-10-29:
加一

Dmitry422, 2024-11-15:
Hello! The same problem ((( "The client Path is invalid"

Dmitry422, 2024-11-15:
I find solution.
Client path must be path to pm3 console client. Something like "/usr/local/bin/pm3/"

Y1ng0, 2024-11-15:
> I find solution. Client path must be path to pm3 console client. Something like "/usr/local/bin/pm3/"
> 
> ![Снимок экрана 2024-11-15 в 12 18 53](https://private-user-images.githubusercontent.com/118886774/386467764-64427697-663d-49b4-ba78-8f22e64080de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE2NjcwODgsIm5iZiI6MTczMTY2Njc4OCwicGF0aCI6Ii8xMTg4ODY3NzQvMzg2NDY3NzY0LTY0NDI3Njk3LTY2M2QtNDliNC1iYTc4LThmMjJlNjQwODBkZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExNVQxMDMzMDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OGYzNGU2NDY0NWFlYTdkMGI3ZmRlNTRiNjYwODZlNDU4NDUwNzg5Y2I2MGFhMTM1YjQ0YWVlNjVhMTVmOGI3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.a_gsXvmi4q7kdBBxZXXuyOI8bxQrp5Rpe06ghJJnN7g)

Can you connect normally?
[image]