wh201906/Proxmark3GUI #68: need update

AVDeveloppement, 2025-05-21:
Hello,
it don't work anymore with last iceman version
is someone can update it please
regards

schmiernippel, 2025-05-25:
I dont work for me also :-( Please update, its such a nice program :-) Thanks alot!!!

Inzhyner, 2025-05-26:
Try replacing the executable file with the attached one [Proxmark3GUI.zip](https://github.com/user-attachments/files/20445296/Proxmark3GUI.zip)

AVDeveloppement, 2025-06-01:
Hello, i tried and i have this error now:

> ---------------------------
> proxmark3.exe - Erreur système
> ---------------------------
> Impossible d’exécuter le code, car libgcc_s_seh-1.dll est introuvable. La réinstallation du programme peut corriger ce problème. 
> ---------------------------
> OK   
> ---------------------------
> ---------------------------
> proxmark3.exe - Erreur système
> ---------------------------
> Impossible d’exécuter le code, car libgd.dll est introuvable. La réinstallation du programme peut corriger ce problème. 
> ---------------------------
> OK   
> ---------------------------
> ---------------------------
> proxmark3.exe - Erreur système
> ---------------------------
> Impossible d’exécuter le code, car libbz2-1.dll est introuvable. La réinstallation du programme peut corriger ce problème. 
> ---------------------------
> OK   
> ---------------------------
> ---------------------------
> proxmark3.exe - Erreur système
> ---------------------------
> Impossible d’exécuter le code, car libjansson-4.dll est introuvable. La réinstallation du programme peut corriger ce problème. 
> ---------------------------
> OK   
> ---------------------------
>